mike-matera/ArduinoSTL #1: Error compiling for board SparkFun Pro Micro.

freemanjiang, 2016-08-10:
Arduino: 1.6.9 (Windows 7), Board: "SparkFun Pro Micro, ATmega32U4 (5V, 16 MHz)"

C:\Users\jy\Documents\Arduino\libraries\ArduinoSTL\src\ArduinoSTL.cpp:9:23: error: 'Serial' was not declared in this scope

   ohserialstream cout(Serial);

                       ^

C:\Users\jy\Documents\Arduino\libraries\ArduinoSTL\src\ArduinoSTL.cpp:10:22: error: 'Serial' was not declared in this scope

   ihserialstream cin(Serial);

                      ^

C:\Users\jy\Documents\Arduino\libraries\ArduinoSTL\src\ArduinoSTL.cpp:28:46: error: 'Serial' was not declared in this scope

 static __Arduino_STDIO_hack __stdio_wrapper(&Serial);

                                              ^

exit status 1
Error compiling for board SparkFun Pro Micro.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

mike-matera, 2016-08-10:
Hi! Thanks for the report. This platform had the Serial port declared differently than others. I'm looking into how to fix it. Stay tuned!

mike-matera, 2016-08-10:
@freemanjiang 

Thanks for the report. I've patched the problem and updated to version v0.1.2. The new version compiles properly for the SparkFun Pro Micro. I don't have one to test. However, I did test the new version on Teensy 2.0 (which is also a 32u4) and it works.

closed 2016-08-10 by mike-matera