mike-matera/ArduinoSTL #2: Not working for a MKR1000

apuder, 2016-08-23:
I'm getting the error message below when compiling for the MKR1000. Is this something that can easily be fixed or is there a bigger issue here?

Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino MKR1000"

Build options changed, rebuilding all
WARNING: library ArduinoSTL claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [samd] architecture(s).
/Users/arno/Documents/Arduino/libraries/ArduinoSTL/src/ArduinoSTL.cpp: In constructor '__Arduino_STDIO_hack::__Arduino_STDIO_hack(Stream*)':
/Users/arno/Documents/Arduino/libraries/ArduinoSTL/src/ArduinoSTL.cpp:60:44: error: 'fdevopen' was not declared in this scope
   fdevopen(arduino_putchar, arduino_getchar); 
                                            ^
exit status 1
Error compiling for board Arduino/Genuino MKR1000.

mike-matera, 2016-08-23:
I've updated the code to avoid compiling AVR specific linkage for printf(). The only non-AVR platform I have to test with is Teensy 3.0. If you try to use printf() it causes a compile error. However, cout worked. I don't have an MKR1000. Try downloading the library and testing it. If it works I'll tag the release.

apuder, 2016-08-23:
Your patch fixes the compile error. Thanks! There is still this warning though:
Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino MKR1000"

Build options changed, rebuilding all
WARNING: library ArduinoSTL claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [samd] architecture(s).

Sketch uses 9,512 bytes (3%) of program storage space. Maximum is 262,144 bytes.

mike-matera, 2016-08-23:
Would you test if cout and cin work? 

Also, I saw this morning that using printf() didn't cause an error on your platform, would you check to see if it works? 

If you can confirm they work I'll add "samd" to the list of supported platforms and make a new release. That will fix the warning.

apuder, 2016-08-25:
cout works perfectly fine. I think you can safely add it to the supported platforms.

closed 2016-08-30 by mike-matera