mike-matera/ArduinoSTL #5: Build failing on Arduino 1.8.1 Ubuntu for the intel edison

xophere, 2017-03-05:
In file included from /home/xophere/Arduino/libraries/ArduinoSTL/src/eh_alloc.cpp:25:0:
/home/xophere/Arduino/libraries/ArduinoSTL/src/unwind-cxx.h:41:20: fatal error: unwind.h: No such file or directory
 #include "unwind.h"

mike-matera, 2017-05-24:
Sorry for the delay... I'll take a look at this.

mike-matera, 2017-05-24:
Hi @xophere. I've done some investigation and I'm not going to fix this bug. There's a good reason: the STL is already present in the Edison libraries, all you have to do is use them. The bundled STL is certain to be superior to uClibc++, which is meant to enable platforms where there's no native STL. Here's code that worked for me:

```C++

#include <vector>
#include <iostream>

using namespace std; 

void setup() {
  Serial.println("Hello World");
}

void loop() {
  vector<int> myvector(10);
  myvector[0] = 1; 
  cout << myvector[0] << endl;
}
```
NOTE: I don't have an Edison so I can't try this code. I have no idea what `cout` really does. It's possible that it does nothing useful unless you adapt it somehow. Serial ports are a bit different than a Linux filehandle. Look in `ArduinoSTL/src/serstream` to see how that's done in ArduinoSTL. 

I'd appreciate it if you would be so kind as to let me know how it goes.

closed 2017-05-24 by mike-matera

xophere, 2017-05-24:
Sorry I am no longer working with the edison.  Hopefully your clarification
is correct.