mike-matera/ArduinoSTL #13: printf blocks for a long time when serial monitor gets disconnected

iFreilicht, 2017-08-10:
To reproduce, connect an LED to pin 13 (if one isn't already), copy [this file](https://gist.github.com/iFreilicht/68830ad183564bc10920db33836927e2#file-arduino_stl_printf_blocking-ino) to a new sketch and upload the code to your Arduino.

Then, watch the LED blink. If you then open the serial monitor, you can see "running... " being printed periodically.

If you then close the serial monitor again, the LED blinks very slowly. Instead of 1000ms for one full on-off cycle, it takes about 6500ms. When turning the serial monitor back on, speed reverts to normal.

The same thing happens when using Arduino's built-in `Serial.println` [like this](https://gist.github.com/iFreilicht/68830ad183564bc10920db33836927e2#file-serial_println_blocking-ino), however the total duration only increases to about 2000ms.

Is this fixable? Can I maybe try and detect a disconnected serial stream to work around this?

mike-matera, 2017-08-10:
What board are you using? 

I was not able to reproduce the problem on an Uno. I suspect that you may have an Arduino with an integrated USB controller like an ATMega32u4 that's on the nano and many others. The UARTS on my ATMega328P just blast out the bits whether or not anyone is listening and I don't see any change in cycle time. I measured with millis() as well as looked. 

However, if the problem also exists (to a different degree) when using Serial.print() there's probably nothing I can do about it. The printf() function is implemented using Serial.write(). Here's the code that printf uses from ArduinoSTL.cpp

```C++
static int arduino_putchar(char c, FILE* f) {
  Stream *uart = __stdio_wrapper.getUart();
  if (c == '\n') uart->write('\r');
  return uart->write(c) == 1? 0 : 1;
}
```

As you can see I call write for every character --because that's what printf needs-- which isn't particularly efficient. That might explain why printf() has a greater delay.

iFreilicht, 2017-08-10:
Yes that's right, I'm using an Arduino Micro. 

> The printf() function is implemented using Serial.write(). 

This explains why it seemed to completely lock the whole device up in my production code (where much more text is being printed).

What seems to work is to count how long a printf takes and if it takes too long restart the program like so:

```c++
int tic = millis();
printf("running... ");
if (millis() - tic > 500){
  asm volatile("jmp 0");
}
```

Thanks for the quick answer, I think you can close this issue as unfixable.

mike-matera, 2017-08-10:
This might work too: 

https://www.arduino.cc/en/Serial/IfSerial

It looks like you can determine if the serial port is attached. Try this instead: 
```C++
if (Serial) {
  printf("running...");
}
```
The documentation doesn't make it clear if this should only be used to time initialization or to more generally detect the presence of a serial monitor.

closed 2017-08-10 by mike-matera

iFreilicht, 2017-08-11:
Ok I tried this, but it doesn't seem to be working. `if(Serial)` is still true after the serial Monitor disconnected. Not to worry, thanks for your help!