mike-matera/ArduinoSTL #15: Including <initializer_list> with library

tttapa, 2017-08-18:
Hi, 
I'm currently working on an Arduino library and I would like to use initializer lists to make it easier to use. This is supported on Arduino-compatible boards with ARM, ARM/SAM or Xtensa architectures, but not on AVR.
I'm able to solve this problem by including your [initializer_list](https://github.com/mike-matera/ArduinoSTL/blob/master/src/initializer_list) file. I just copied it to the src folder of my own library as _initializer_list.h_ and added the following to my library source files:
```
#ifdef __AVR__
#include "initializer_list.h"
#else
#include <initializer_list>
#endif
```
Everything works great, however, I have no idea if it's allowed to redistribute this file with my library, and if so, how to do it without infringing any copyright laws.

Thanks in advance,
Pieter

mike-matera, 2017-08-18:
Thanks for reaching out. Does your library have a software license? Is it open source? 

Cheers
./m

tttapa, 2017-08-18:
Yes, the library is open source, I just chose one of the default licences on GitHub, it's a GNU General Public License v3.0.
[https://github.com/tttapa/MIDI_controller/blob/master/LICENSE](https://github.com/tttapa/MIDI_controller/blob/master/LICENSE)

Thank you, 
Pieter

mike-matera, 2017-08-18:
Great. Help yourself! Thanks for checking in.

closed 2017-08-18 by mike-matera

tttapa, 2017-08-18:
Thank you very much!