mike-matera/ArduinoSTL #18: Facing problem with usase of functional

arihantdaga, 2018-01-09:
`
typedef std::function<void(uint8_t pin, uint8_t event, uint8_t count, uint16_t length)> TDebounceEventCallback;

`

I have a library which is using function templates like this, and i am little new with this type of code in C++. But i found that in the arduinoSTL library there is no template for function instead only for unary_function and binary_function... 
Can somebody help me here how can i use function of that sort...

mike-matera, 2018-01-11:
I'm not very familiar with the functional code. What I can tell you is that the STL implementation comes from here: 
https://cxx.uclibc.org/index.html
I have just done a little work to make is easy to use in Arduino and added some streams support. The upstream project hasn't had a revision in a while. I'm always happy to take PRs.

arihantdaga, 2018-01-12:
@mike-matera  Thanks anyway...

cpk1971, 2018-03-26:
another C++11 thingy that makes use of lambdas a lot more friendly

it's not 100% necessary--you can still use function pointers...but function<> makes it agnostic between lambdas, function pointers, overloads of () and so on

closed 2022-01-03 by mike-matera