mike-matera/ArduinoSTL #20: Client side Serial binding to cin/cout

neu-rah, 2018-03-31:
Hi!

Saw your code comments about it, and yes its needed...

my approach was to remove the .cpp implementation
```c++
namespace std
{
  // ohserialstream cout(Serial);
  // ihserialstream cin(Serial);
}
```

and add this to  my sketch

```c++
ohserialstream std::cout(Serial);
ihserialstream std::cin(Serial);
```

for me it sounds as a nice trade-off, no?

neu-rah, 2018-03-31:
took it a step further and renamed your hack class for something more user friendly (but the name is not he point) and let cin/cout depend on the stdio hack, and only that needed to be specified on sketch...
i would go a bit further and use a reference instead of pointer (more user friendly).

```c++
#include <ArduinoSTL.h>
using namespace std;

const ArduinoSTDIO stdioUse(&Serial);

void setup() {
  Serial.begin(115200);
  while(!Serial);
  cout<<"STDIO on Arduino"<<endl;
}

void loop() {}
```

is this an acceptable trade off?

mike-matera, 2018-04-02:
Thanks! Let me try it out and think about it a bit.

mike-matera, 2018-04-03:
Quick question. Are you doing this so you can bind to a different serial port? 

I need something that is backwards compatible. I don't want people to download the newest version of the library and find that it breaks sketches that once worked. So if you need to switch serial ports there would have to be some extra steps to prevent the default binding.

neu-rah, 2018-04-03:
yes it was to bind to a different serial port, and yes it not only breaks compatibility as it forces the user to point one by presenting an error message if not...
so its not an easy trade off, but it allows me to set the port, is there other way?

neu-rah, 2018-04-03:
of course i can test it
a predefined list was crossing my mind because we have some system identifiers, but on some systems there are ambiguity has multiple ports are available. Its unfortunate we cant check if a variable is defined and provide a default value if not

mike-matera, 2018-04-04:
Great. I've created a branch for you test. Look at the changes in PR #21 . There are instructions for how to provide your own serial device in the PR and in the README.md in that branch. 

It works for me. Please let me know if you get it working.

mike-matera, 2018-04-04:
Fixed!

closed 2018-04-04 by mike-matera