mike-matera/ArduinoSTL #25: Issues using map

radsolution, 2018-11-21:
Hi Mike.
I'm using your library, and everything seems to work except map.
Here's my declaration in a .h file:
`std::map<int, edSession*> sessions;`
and the error I get is:
`this declaration has no storage class or type specifier`
What is wrong, please?
I also can't see how string, map and vector get pulled into my .h file when I add:
`#include <ArduinoSTL.h>`
Can you please explain? I'm fairly new to Arduino/C/C++

mike-matera, 2018-12-07:
Hello! 

Don't forget to include `map`. Here's a sketch that works: 

```c++
#include <ArduinoSTL.h>
#include <map>

std::map<std::string, std::string> m; 

void setup() {
  m["hello"] = "world";
}

void loop() {
  std::cout << m["hello"] << std::endl;
}
```

closed 2018-12-07 by mike-matera

radsolution, 2018-12-10:
Thanks Mike