mike-matera/ArduinoSTL #28: width() non-functional (for me)

dougp2, 2019-01-12:
This code:
[code]
// width.cpp -- using the width method
 // C++ primer 17.4, page 1080

#include <ArduinoSTL.h>
//#include <PrintStream.h>

void setup() {
 
Serial.begin(115200);

    using std::cout;
    int w = cout.width(30);
    cout << "default field width = " << w << ":\n";

    cout.width(5);
    cout << "N" <<':';
    cout.width(8);
    cout << "N * N" << ":\n";

    for (long i = 1; i <= 100; i *= 10)
    {
        cout.width(5);
        cout << i <<':';
        cout.width(8);
        cout << i * i << ":\n";
    }
    // std::cin.get();
  
}[/code]

void loop() {}

is supposed to print right justified.  It doesn't, everything on a line is jammed together.

Am I doing something wrong?

kirkins, 2019-01-19:
I'm having a similar issue with accumulate:

    'accumulate' is not a member of 'std'

Running as:

    float average = std::accumulate(avgTemp.begin(), avgTemp.end(), 0.0)/avgTemp.size();

mike-matera, 2022-01-03:
This seems to be a limitation of the underlying uClibc++ library. It looks like some of the functionality is implemented but it clearly doesn't work.