mike-matera/ArduinoSTL #29: <array>?

lf-, 2019-02-25:
I see that there's valarray but not array. This strikes me as strange because it seems like array would be the logical choice to use on a microcontroller for static data, which is what I want array for. 

Use case: constant array of instructions to run on some given input, compatible with c++ features and avoiding using a c array.

dasfdlinux, 2019-07-18:
> Use case: constant array of instructions to run on some given input, compatible with c++ features and avoiding using a c array.

For instance, I'm looking to make a contexpr. However, std::vector can't be a constexpr, std::array doesn't exist, and the built in array doesn't have a size() function for use later in my code.

lf-, 2019-07-21:
Precisely. Unfortunately this is not a C++11 implementation, which is why array is missing. Now, there is an implementation of C++14 (17?) basic STL features at https://github.com/ambroise-leclerc/ETL, but I was having issues making my build environment work with it (which is totally inexperience at work).

ciband, 2019-09-27:
Added #38 to resolve this

mike-matera, 2022-01-03:
#38 was merged. Thanks @ciband

closed 2022-01-03 by mike-matera