mike-matera/ArduinoSTL #31: Will not compile for Teensy 3.1/3.2

systemofapwne, 2019-06-26:
I struggled the past days to compile my project, where other libs included ArduinoSTL, even though, they did not use it. It took me some time to figure it out. The following code will not compile on PlatformIO@VSCode.

This is my basic code:

https://pastebin.com/7KE4puMr

The Error is  https://pastebin.com/vm51i2zQ

It looks like, C++11 keywords are not recognized, even though, everything is compiled with at least C++11, if not C++14. I tested this on ArduinoIDE too (with teensyduino-core) and had similar troubles.

Solution to my problem: Exclude ArduinoSTL from compile. But this won't fix the incompatibility of this lib with the compile.

Note: ArduinoSTL compiles perfectly fine on Arduino Uno/Nano though.

mike-matera, 2019-06-27:
Hello! You don't need ArduinoSTL on ARM platforms. The compiler comes with the STL on those. You can just include your favorite library!

closed 2019-06-27 by mike-matera

systemofapwne, 2019-06-27:
Hello,
Thank you for your reply. This explains, why my code compiles, even though I had excluded ArduinoSTL :D. I anyway suggest to add an include-guard, if anything else but the AVR platform is present OR if the STL is present, since this causes a lot of trouble, which is not immediately clear, where the cause is.
The reason is, that the libraries (one written by my own) do not explicitly include any STL headers but e.g. uses "std functions" like memcpy here and there. This causes the compiler to actually include ArduinoSTL on its own, even though when on an ARM platform, it won't be necessary. I totally would address this bug to the direction of teensyduino or arduino core itself. Yet, adding an include-guard in ArduinoSTL might also do the trick.

maikkel, 2021-02-03:
> Hello! You don't need ArduinoSTL on ARM platforms. The compiler comes with the STL on those. You can just include your favorite library!

Is that true? I can't include any std functions on the teensy 3.6, and Arduino STL only throws errors.