mike-matera/ArduinoSTL #32: ESP-32 support?

calchuchesta, 2019-07-20:
Due to some issues on my current project, I've decided to switch from an Arduino Nano to an ESP-32, however I'm getting these errors whenever compiling my code:

`
WARNING: library ArduinoSTL-master claims to run on (avr, samd) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
In file included from sketch\ino.ino.cpp:1:0:

C:\Users\azedo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:163:12: error: 'std::isinf' has not been declared

 using std::isinf;

            ^

C:\Users\azedo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32/Arduino.h:164:12: error: 'std::isnan' has not been declared

 using std::isnan;

exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.
`

neu-rah, 2019-07-20:
esp32 has its own STL, so you should not include this one (`ArduinoSTL.h`)

SamsTheNerd, 2019-12-26:
what would you include instead? to get features like vectors and strings?

CristianVladescu, 2019-12-27:
> what would you include instead? to get features like vectors and strings?
> Edit: nvm, I just included <vector> and <string> and it worked. I also have this library as well as the [StandardCPlusPlus library](https://github.com/maniacbug/StandardCplusplus) downloaded so I'm not sure which one did it, for any future readers

Neither of them did it. The vector was included from esp32 toolchain:
[image]

closed 2022-01-03 by mike-matera