mike-matera/ArduinoSTL #33: ArduinoSTL and Visual Micro

yet-another-average-joe, 2019-07-26:
Hello,

I recently decided to code for the STM32s I got a couple of years ago. And indeed, I decided to use the STL... (on a Black Pill)

But, is STM32 supported ?

In Arduino IDE : it seems it is.

In Visual Studio + Visual Micro, your library is not recognized, and headers not included until "architectures" is set to "*" in file library properties. And even with this change, I immediatly ran into problems (Visual Micro only). I don't understand why... They share the same gcc.

I found nothing on the web about such an issue, and STM32duino website is down (Server quota exceeded).

Here's some code that compiles and run fine with Arduino IDE, but not Visual Micro :

```
#include <ArduinoSTL.h>

class Item
{
	public:
	Item(uint32_t i, std::string s): index(i), str(s) {}
	std::string str;
	uint32_t index;
};

std::vector<Item> Vector;
Item item0(5, std::string("five"));
Item item1(4, std::string("four"));
Item item2(3, std::string("three"));
Item item3(2, std::string("two"));
Item item4(1, std::string("one"));
Item item5(0, std::string("zero"));

bool _compare(Item& item1, Item& item2)
{
	return item1.index < item2.index;
}

void setup()
{
	Serial.begin(9600);
	while (!Serial) ;
	Serial.print("hello\n\n");
	Vector.push_back(item0);
	Vector.push_back(item1);
	Vector.push_back(item2);
	Vector.push_back(item3);
	Vector.push_back(item4);
	Vector.push_back(item5);

for (auto& iter : Vector)
   Serial.println(iter.str.c_str());

Serial.println("\nsort :\n");

std::sort(Vector.begin(), Vector.end(), compare);

for (auto& iter : Vector)
   Serial.println(iter.str.c_str());
}

void loop()
{
}
```

Errors in Visual Micro ; seems to be related to std::sort()

```

TestSTL.ino: 4:15: error: 'Item' was not declared in this scope
 
TestSTL.ino: 4:21: error: 'item1' was not declared in this scope
 
TestSTL.ino: 4:28: error: 'Item' was not declared in this scope
 
TestSTL.ino: 4:34: error: 'item2' was not declared in this scope
 
TestSTL.ino: 4:39: error: expression list treated as compound expression in initializer [-fpermissive]
TestSTL.ino: In constructor Item::Item(uint32_t, std::string)
TestSTL.ino: 9:11: warning: 'Item::index' will be initialized after [-Wreorder]
   uint32_t index
TestSTL.ino: 8:14: warning:   'std::string Item::str' [-Wreorder]
Error compiling project sources
  std*: string str
TestSTL.ino: 6:2: warning:   when initialized here [-Wreorder]
  Item(uint32_t i, std*: string s): index(i), str(s) {}
 
TestSTL.ino: In function bool _compare(Item&, Item&)
 
TestSTL.ino: 20:39: error: 'bool _compare(Item&, Item&)' redeclared as different kind of symbol
   bool _compare(Item& item1, Item& item2)
 
TestSTL.ino: 4:6: error: previous declaration of 'bool _compare
 
TestSTL.ino: In function void setup()
 
TestSTL.ino: 43:42: error: 'compare' was not declared in this scope
  std*: sort(Vector.begin(), Vector.end(), compare)
Build failed for project 'TestSTL'

```

yet-another-average-joe, 2019-07-27:
I finally posted a bug report on the Visual Micro forum.

There's a workaround to this problem : declaring the comparison function as an operator in the "Item" class, 

```
	bool operator <(const Item& item) const
	{
		return index < item.index;
	}

```

and calling sort() with no comparison function.

`	
       std::sort(Vector.begin(), Vector.end());
`

do work fine.

neu-rah, 2019-07-27:
I think this library is intended for AVR's, STM's like ESP's should already have they own STL, no need to include this one.
Not sure if its the cause but the test is easy just remove the `#include <ArduinoSTL.h>`

yet-another-average-joe, 2019-07-27:
You are right.

But I use VisualStudio+Visual Micro, and this makes a huge difference. When using the integrated STL, Intellisense  (code completion) does not work with the template classes.

Even worse, the integrated STL kills Intellisense for the whole code, STL related or not. I don't know why, and I'm not interested. From my point of view, the integrated STL is just unusable !

Intellisense works fine with ArduinoSTL and causes no problem. It is 100% Visual Studio / Intellisense friendly.

Maybe you don't know Visual Studio (for example if you're a Linux guy).

Suppose a deque or whatever is declared. When typing the name of the deque followed by a '.' or '->', a listbox appears, with all functions and operators. After a function is selected, another list appears with the different variants of the function (parameters and types). Intellisense is clever enough to list the parameters with the variable types the deque was defined with. This makes coding extremely efficient, quick, and intuitive. Could'nt live without. (I have been using Visual C++, then Visual Studio since the 1st beta I received from the MSDN in 1993 ; Intellisense came out in 1997 ; I've been maried with for a long time, and it's a love story ; VS is one of the 2 softwares why I stick with Windows)

Maybe the integrated STL could work with VisualStudioCode + PlatformIO, but I hate this thing. No floating windows, popups everywhere, confusing interface, not multi monitor friendly, etc.

Therefore : ArduinoSTL or no STL. Period !

Maybe I could help to test the library with STM32duino ? And maybe add some functionalities that are missing ? (for example, it seems that collection initialisation from a operator= {array} is missing)

@neu-rah (off topic) : began to learn about your menu library a couple days ago... At first I wanted to use ArduinoSTL for creating dynamic menus.

NameOfTheDragon, 2019-08-04:
I'm having similar issues with the MKR-1010 WiFI (a SAMD chip). I notice in the library manager that if I check "Only show libraries supported by the current board" then ArduinoSTL is not shown in the list, so it looks like that's a non-starter.

How does one use the "built in" STL?

yet-another-average-joe, 2019-08-05:
> How does one use the "built in" STL?

just #include the template classes, as usual.

yet-another-average-joe, 2019-08-27:
Some more (weird !) information.

As I wrote, ArduinoSTL seems to work fine with STM32duino at least for collection classes. It could be another story for streams...

But there are some weird issues with STL and VisualMicro. Due to the include paths set by Visual Studio, Microsoft STL classes are used if ArduinoSTL is not installed ! Same problem with Arduino IDE and Visual Studio. No effect after editing these paths.

Therefore, I'm not sure STM32duino comes with an STL variant. But if Visual Studio +  Micro is installed, hex files are 50KB larger when ArduinoSTL library is not used !!!

closed 2022-01-03 by mike-matera