mike-matera/ArduinoSTL #43: cout should support const char arrays from flash

pischky, 2019-09-30:
It should be possible to use Macro "F()" to store strings (const char*) from flash memory (PROGMEM) like Arduinio print/println does. This saves RAM.

`cout << F("\r\nSystem halted!") << endl;`

The Arduino library uses class __FlashStringHelper in macro F(). I have added (in "ostream")

````
template<class charT, class traits> _UCXXEXPORT basic_ostream<charT,traits>&
    operator<<(basic_ostream<charT,traits>& out, const __FlashStringHelper* c)
````
and
````
template<class traits> _UCXXEXPORT basic_ostream<char,traits>&
    operator<<(basic_ostream<char,traits>& out, const __FlashStringHelper* c)
````
With code from Arduino library file Print.cpp. Have given it a short test and it works on Arduino UNO with current IDE / eclipse.

Will try to attach patch.

pischky, 2019-09-30:
[ostream.patch.zip](https://github.com/mike-matera/ArduinoSTL/files/3670802/ostream.patch.zip)

This zip contains patch file and complete changed file "ostream". Feel free to remove "[PI: New]"

BTW: Thanks for this library. Makes life much easer ;-)

chrisco484, 2021-01-16:
> [ostream.patch.zip](https://github.com/mike-matera/ArduinoSTL/files/3670802/ostream.patch.zip)
> 
> This zip contains patch file and complete changed file "ostream". Feel free to remove "[PI: New]"
> 
> BTW: Thanks for this library. Makes life much easer ;-)

Thanks! This is exactly what I needed as my free mem on a 328P got very low and the code was using AduinoSTL with lots of 

`cout << "blah" << endl;`

chrisco484, 2021-01-16:
> [ostream.patch.zip](https://github.com/mike-matera/ArduinoSTL/files/3670802/ostream.patch.zip)
> 
> This zip contains patch file and complete changed file "ostream". Feel free to remove "[PI: New]"
> 
> BTW: Thanks for this library. Makes life much easer ;-)

I've just tested this patch and it works brilliantly. I've been able to save a lot of precious RAM on a few "third drawer" Uno borads I have lying around.

Have you raised a PR for this patch? A PR it will be more likely to be merged into master.

mike-matera, 2022-01-03:
@chrisco484 Would you send me this as a PR? I'd be happy to take it.

chrisco484, 2022-01-04:
> @chrisco484 Would you send me this as a PR? I'd be happy to take it.

Hi Mike,

I ended up hitting too many resource constraints with 328P based boards so moved up to STM32 blue pills/black pills with ARM Cortex CPUs - wow! So much faster and so much more resources and actually cheaper and they still support the Arduino "APIs" so all my Arduino software targeting the 328Ps just recompiled with very little change required. So it was a no brainer.

I'll try to find some time to convert @pischky 's patch zip to a PR.

pischky, 2022-01-04:
Hello,

I have created a Pull Request. Did not know that you can edit online
on github without local git installation. See #69
Thanks for your library.

Martin