mike-matera/ArduinoSTL #53: Memory usage of this library

alve89, 2020-03-20:
This library uses 22 % (468 or 459 bytes) of the dynamic memory (RAM). Since I have trouble with memory with one of my projects, is there a way to reduce your library's memory usage?

matthijskooijman, 2020-09-17:
I guess it really depends on what parts of the library you use, I would suspect that just including the library would not really add any memory by itself. It's also not unlikely that in comparing the usage "with" and "without" this library, you might have accidentally left out other things and are including memory used by parts of the Arduino core in your comparison.

I'm not sure if this is even the right platform for this question, but if you want any chance of a meaningful response, you should probably post at least a minimal sketch that shows what parts of the library you use and shows how you compare the memory with and without. Also make sure to note the board you are using.

mike-matera, 2022-01-03:
Hi! I'm afraid there's no good way to reduce memory. If you're in a memory constrained environment you should avoid using the STL.

closed 2022-01-03 by mike-matera