mike-matera/ArduinoSTL #56: Trivial sketch won't compile using Arduino 1.8.13

nholthaus, 2020-06-20:
This library worked mostly fine for me in Arduino 1.8.12, but in 1.8.13 the following trivial sketch won't compile. My intuition is that it's an include-order dependency issue.

```cpp
#include <ArduinoSTL.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

with the following errors:

> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\del_opvs.cpp:25:53: error: 'std::size_t' has not been declared
> 
>  _UCXXEXPORT void operator delete[](void * ptr, std::size_t) throw(){
> 
>                                                      ^~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\del_opnt.cpp:25:56: error: 'nothrow_t' in namespace 'std' does not name a type
> 
>  _UCXXEXPORT void operator delete(void* ptr, const std::nothrow_t& ) throw() {
> 
>                                                         ^~~~~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\del_ops.cpp:25:50: error: 'std::size_t' has not been declared
> 
>  _UCXXEXPORT void operator delete(void* ptr, std::size_t) throw(){
> 
>                                                   ^~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\del_opvnt.cpp:25:58: error: 'nothrow_t' in namespace 'std' does not name a type
> 
>  _UCXXEXPORT void operator delete[](void* ptr, const std::nothrow_t& ) throw(){
> 
>                                                           ^~~~~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_opnt.cpp:25:37: error: declaration of 'operator new' as non-function
> 
>  _UCXXEXPORT void* operator new(std::size_t numBytes, const std::nothrow_t& ) throw(){
> 
>                                      ^~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_opnt.cpp:25:37: error: 'size_t' is not a member of 'std'
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_opnt.cpp:25:37: note: suggested alternative:
> 
> In file included from c:\users\nholt\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\stdlib.h:48:0,
> 
>                  from C:\Users\NHolt\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/new.h:22,
> 
>                  from C:\Users\NHolt\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.3\cores\arduino/new:5,
> 
>                  from E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_opnt.cpp:20:
> 
> c:\users\nholt\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\lib\gcc\avr\7.3.0\include\stddef.h:216:23: note:   'size_t'
> 
>  typedef __SIZE_TYPE__ size_t;
> 
>                        ^~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_opnt.cpp:25:54: error: expected primary-expression before 'const'
> 
>  _UCXXEXPORT void* operator new(std::size_t numBytes, const std::nothrow_t& ) throw(){
> 
>                                                       ^~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_handler.cpp:22:12: error: 'nothrow_t' in namespace 'std' does not name a type
> 
>  const std::nothrow_t std::nothrow = { };
> 
>             ^~~~~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_handler.cpp:25:6: error: 'new_handler' in namespace 'std' does not name a type
> 
>  std::new_handler __new_handler;
> 
>       ^~~~~~~~~~~
> 
> E:\Users\NHolt\Documents\Arduino\libraries\ArduinoSTL-master\src\new_handler.cpp:27:1: error: '_UCXXEXPORT' does not name a type
> 
>  _UCXXEXPORT std::new_handler std::set_new_handler(std::new_handler new_p) throw(){
> 
>  ^~~~~~~~~~~
> 
> exit status 1

nholthaus, 2020-06-20:
The problem appears to be that if you have other IDEs installed (e.g. Visual Studio), the include search paths find the platforms `std` library implementation headers. Switching the include statements from `<>` to `""` forces the use of the local `std` library implementation, which is what you want here. I'll submit a patch.

roelschroeven, 2020-06-23:
I think there's another reason: new versions of the AVR core have a header file 'new' that conflicts with the one from ArduinoSTL (https://github.com/arduino/ArduinoCore-avr/commit/d1ae194f8d82027cd6cbe35e32111550c9bba9fd). If you open the Arduino IDE, go to Tools -> Board -> Board manager and update the Arduino AVR Boards to version 1.8.3 you'll likely have the same problem.

The result is the same, so your pull request should solve it all the same.

Workaround in the meantime: downgrade Arduino AVR Boards to 1.8.2.

mike-matera, 2020-07-18:
Hi. I'll take a look at fixing this. I haven't used Arduino in a while.

Dantali0n, 2020-09-17:
I can confirm I am experiencing the same problems as @roelschroeven using IDE 1.8.3 for AVR devices

matthijskooijman, 2020-09-17:
This is indeed a problem triggered by Arduino adding a `<new>` header that is not actually complete. It already offered it as `new.h` and it seemed like a good idea to offer it as `<new>` as well, but I hadn't considered this problem when I reviewed that PR :-)

I believe the proper solution on the Arduino end is to make sure that it implements the `<new>` header completely and in a standards-compliant way, I believe this should make ArduinoSTL work again as well. I'm planning to provide a PR to Arduino for this, see https://github.com/arduino/ArduinoCore-avr/issues/287 for some additional discussion of that.

matthijskooijman, 2020-09-17:
I created a PR for the Arduino AVR core, testing is welcome: https://github.com/arduino/ArduinoCore-avr/pull/361

That *almost* solves this issue, the one remaining problem I saw is that `src/new_handler.cpp` in ArduinoSTL only includes `<new>`, and assumes that that makes `_UCXXEXPORT` available, However, when the Arduino version of `<new>` is used, this assumption is incorrect. For other cpp files that include `new` this is not an issue because they also include other files.

I don't think this can be fixed on the Arduino side, but I can see two ways to fix this in ArduinoSTL:
 1. Use `#include "new"`, as suggested before (but maybe not for *all* includes, just for `new`?)
 2. Explicitly add `#include <basic_definitions>` to `new_handler.cpp` to make `_UCXXEXPORT` available. I tried this, and that works. This is still a workaround, since it is really only needed because we know that `new` will be provided by Arduino rather than ArduinoSTL.

I'm not sure which would be the best solution, though.

matthijskooijman, 2020-10-15:
The PR in the AVR core is merged, so that leaves just the second issue I noted above.

Dantali0n, 2020-10-15:
> The PR in the AVR core is merged, so that leaves just the second issue I noted above.

Thank you for your efforts at making ArduinoSTL work with AVR boards 1.8.3

matthijskooijman, 2020-11-19:
Hm, I just noticed a new problem:

```
new.cpp.o (symbol from plugin): In function `operator new(unsigned int)':
(.text+0x0): multiple definition of `std::nothrow'
/tmp/arduino_build_fefirmware/libraries/ArduinoSTL/new_handler.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

I'm not sure why this didn't happen to me before, since it quite makes sense that it happens: Both Arduino (https://github.com/arduino/ArduinoCore-avr/blob/60f0d0b125e06dbf57b800192c80e5f60d681438/cores/arduino/new.cpp#L36, since merging https://github.com/arduino/ArduinoCore-avr/pull/361) and ArduinoSTL (https://github.com/mike-matera/ArduinoSTL/blob/7411816e2d8f49d96559dbaa47e327816dde860c/src/new_handler.cpp#L22) declare the `std::nothrow` object. It's just a dummy, empty object, but the linker doesn't know this.

I'm not sure how to solve this, *maybe* it works to make it `constexpr` and declare it in the header rather than a .cpp file (and since it's empty and there is no reason to take the address of the object, it might not be emitted at all then), or maybe it should be an inline variable (but that's C++17)...

MishaRuko, 2020-12-11:
Instead of replacing every occurrence of `#include <new>` with `#include "new"` in the ArduinoSTL library to get the example sketch to compile, as was proposed here: https://github.com/mike-matera/ArduinoSTL/pull/57, a solution that worked for me was to find the AVR boards 1.8.3 source code and change the name of the file called "new" (no extension) to something else. I don't know if this breaks anything else but it seems to work fine.

(On MacOS the file is /Users/user/Library/Arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino/new)

fl0wm0ti0n, 2020-12-11:
Hello i run into the same problem, just found this lib to use vectors...

i use Visual Studio with Visual Micro and the Arduino IDE V1.8.13 in the background. i have tried include with <> and with "" doesnt matter...

Thanks for any Help!!

Schweregrad	Code	Beschreibung	Projekt	Datei	Zeile	Unterdrückungszustand
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	40	
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	46	
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	53	
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	56	
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	61	
Fehler (aktiv)	E0351	Der erste Parameter der Speicherbelegungsfunktion muss vom Typ "size_t" sein.	JuliansAnhaengerLedControll	C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new	64	
Fehler		25:56: error: 'nothrow_t' in namespace 'std' does not name a type		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\del_opnt.cpp	25	
Fehler		25:50: error: 'std::size_t' has not been declared		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\del_ops.cpp	25	
Fehler		25:58: error: 'nothrow_t' in namespace 'std' does not name a type		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\del_opvnt.cpp	25	
Fehler		25:53: error: 'std::size_t' has not been declared		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\del_opvs.cpp	25	
Fehler		22:12: error: 'nothrow_t' in namespace 'std' does not name a type		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_handler.cpp	22	
Fehler		25:6: error: 'new_handler' in namespace 'std' does not name a type		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_handler.cpp	25	
Fehler		27:1: error: '_UCXXEXPORT' does not name a type		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_handler.cpp	27	
Fehler		25:54: error: expected primary-expression before 'const		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opnt.cpp	25	
Fehler		25:37: error: 'size_t' is not a member of 'std		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opnt.cpp	25	
Fehler		25:37: error: declaration of 'operator new' as non-function		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opnt.cpp	25	
Fehler		25:56: error: expected primary-expression before 'const		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opvnt.cpp	25	
Fehler		25:39: error: declaration of 'operator new []' as non-function		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opvnt.cpp	25	
Fehler		25:39: error: 'size_t' is not a member of 'std		C:\Program Files (x86)\Arduino\libraries\ArduinoSTL\src\new_opvnt.cpp	25	

[image]

NetworkAndSoftware, 2021-01-01:
> The PR in the AVR core is merged, so that leaves just the second issue I noted above.

Thank you for you work. I was able to build my project by grabbing the master branch of the avr project and making the change to double quotes in new_handler.cpp as you suggested.

Unfortunately, not std::unique_ptr in the Arduino-STL memory include file... I'll have to stay with basic pointers.

MHamidJamil420, 2021-01-20:
> I think there's another reason: new versions of the AVR core have a header file 'new' that conflicts with the one from ArduinoSTL ([arduino/ArduinoCore-avr@d1ae194](https://github.com/arduino/ArduinoCore-avr/commit/d1ae194f8d82027cd6cbe35e32111550c9bba9fd)). If anyone opens the Arduino IDE, goes to Tools -> Board -> Board manager and updates the Arduino AVR Boards to version 1.8.3, he/she will likely have the same problem.
> 
> The result is the same, so nbolthaus' pull request should solve it all the same.
> 
> Workaround in the meantime: downgrade Arduino AVR Boards to 1.8.2.

thanks man after down grading my problum is solved

NameOfTheDragon, 2021-06-15:
> The problem appears to be that if you have other IDEs installed (e.g. Visual Studio), the include search paths find the platforms `std` library implementation headers. Switching the include [...]

The fix suggested here will likely only work for the AVR targets.

If you are using one of the ARM-based Arduinos such as MKR 1010 WiFi, then that uses the ARM toolchain, which has the Standard Template Library built in. In that case you __must__ use `#include <stdlib>` because the `ArduinoSTL` library will not compile at all and is not really needed, as you have a "proper" STL implementation available.

mike-matera, 2022-01-03:
Hello! I've updated the library and it now compiles using the latest Arduino IDE. Sorry to make you all wait!

closed 2022-01-03 by mike-matera

matthijskooijman, 2022-01-10:
I believe this issue is not entirely fixed yet, in particular https://github.com/mike-matera/ArduinoSTL/issues/56#issuecomment-730555065 is unresolved. See also https://github.com/arduino-libraries/Arduino_AVRSTL/pull/2#issuecomment-1008798285 for some more background, and https://github.com/arduino-libraries/Arduino_AVRSTL/pull/8 for a possible fix (hardly tested, so feedback welcome).

technorainbows, 2022-02-02:
> I believe this issue is not entirely fixed yet, in particular [#56 (comment)](https://github.com/mike-matera/ArduinoSTL/issues/56#issuecomment-730555065) is unresolved. See also [arduino-libraries/Arduino_AVRSTL#2 (comment)](https://github.com/arduino-libraries/Arduino_AVRSTL/pull/2#issuecomment-1008798285) for some more background, and [arduino-libraries/Arduino_AVRSTL#8](https://github.com/arduino-libraries/Arduino_AVRSTL/pull/8) for a possible fix (hardly tested, so feedback welcome).

I confirm with @matthijskooijman -- this is not fixed. Only fix so far is downgrading Arduino AVR board to 1.8.2.

brand17, 2022-05-06:
Have the same error with Arduino IDE 2.0.0. Downgrading board to 1.8.2 works

alexshirley, 2023-03-25:
I would also like to follow up, this is not fixed.

matthijskooijman, 2023-03-27:
I've submitted a fix (last year already) for this here: #82