mike-matera/ArduinoSTL #59: Problems compiling

jay-reynolds, 2020-09-10:
From a minimal setup in a new project, including only:  
#include <ArduinoSTL.h>

I'm getting:

` ... error: '::abort' has not been declared
     27    |  using ::abort;
             |          ^~~~~`

etc...  from cstdlib and similarly for all of cmath.

Using:
Arduino 1.8.13
1.8.2 of Arduino SAMD, 32 bits ARM Cortex-M0+ Board (Adafruit ItsyBitsy M4)
On Windows 10

Strange thing is that it was working fine and then at some point later in the day I upgraded versions in the board manager ... tried downgrading but never got things working again.

The other thing I did earlier in the day was install Atmel Studio 7.   Was able to compile after that, though.

I tried using nholthaus' version from issue 56, but had no luck with it, as the issue looked possibly similar to mine.

Thanks in advance for any help with this issue.

matthijskooijman, 2020-09-17:
But if you're using a SAMD board, do you even need ArduinoSTL? Doesn't gcc/Arduino for ARM/SAMD not just ship a builtin libstdc++ and stl? If so, bringing in ArduinoSTL/uclibc++ will probably only complicate matters and maybe even introduce conflicts?

jay-reynolds, 2020-09-17:
Hi Matt.   I'm not entirely certain as I'm not very familiar with programming microcontrollers, but this:

https://www.arduinolibraries.info/architectures/samd

Has an ArduinoSTL link that points to this repo.

matthijskooijman, 2020-09-18:
Hm, it does seem that this library explicitly advertise support for SAMD: https://github.com/mike-matera/ArduinoSTL/blob/7411816e2d8f49d96559dbaa47e327816dde860c/library.properties#L9

Did you try compiling your code *without* including ArduinoSTL? So just using (only) the C++ standard includes?

jay-reynolds, 2020-09-18:
This solves my problem.  Dank je wel!

Resolved per Matthijs's advice.   Closing.

closed 2020-09-18 by jay-reynolds

matthijskooijman, 2020-09-23:
@mike-matera, it seems that ArduinoSTL does not really work on samd, and is also not needed at all since samd has a perfectly working standard library already.

Maybe you should remove `samd` from the supported architectures? While you're there, maybe add `megaavr` to that list? That uses the same avr-gcc toolchain, but is intended for a line of new-style AVR chips (same CPU core, but with some significant changes to the peripherals). I haven't tested this, but I think ArduinoSTL should be similarly useful as on the regular avr core. See https://github.com/arduino/ArduinoCore-megaavr