mike-matera/ArduinoSTL #60: "Stream has not been declared" with PlatformIO and atmelmegaavr@1.3.0

AlexanderAmelkin, 2020-11-14:
When compiling a simple sketch (even an empty one) when ArduinoSTL is added to `libdeps` in `platformio.ini` for an Arduiono Nano Every board (`board=nano_every`), I get the following error:

```
Compiling .pio/build/nano_every/FrameworkArduino/NANO_compat.cpp.o
In file included from .pio/libdeps/nano_every/ArduinoSTL/src/ArduinoSTL.h:12:0,
                 from .pio/libdeps/nano_every/ArduinoSTL/src/ArduinoSTL.cpp:1:
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:307:9: error: '::Stream' has not been declared
 
         ^     
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:309:73: error: 'Stream' does not name a type; did you mean 'fstream'?
 template <class charT, class traits = char_traits<charT>, class Tserial=Stream> class basic_iserialstream;
                                                                         ^~~~~~
                                                                         fstream
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:310:73: error: 'Stream' does not name a type; did you mean 'fstream'?
 template <class charT, class traits = char_traits<charT>, class Tserial=Stream> class basic_oserialstream;
                                                                         ^~~~~~
                                                                         fstream
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:311:73: error: 'Stream' does not name a type; did you mean 'fstream'?
 
                                                                         ^     
                                                                         fstream
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:313:33: error: template argument 3 is invalid
 typedef basic_oserialstream<char> ohserialstream;
                                 ^
.pio/libdeps/nano_every/ArduinoSTL/src/serstream:314:33: error: template argument 3 is invalid
 
                                 ^
```

AlexanderAmelkin, 2020-11-14:
It appears that `#include <Arduino.h>` is missing at the top of `serstream.cpp`. Adding that fixes the build, but then `printf("test\n")` doesn't produce any output although `cout << "test\n";` works fine.

mike-matera, 2022-01-03:
Hi! My latest build fixes the problems with PlatformIO and latest Arduino. Thanks!

closed 2022-01-03 by mike-matera