mike-matera/ArduinoSTL #61: Initializing vector<string> Beyond a Number of Entries Crashes Sketch

rmerriam, 2020-11-17:
Arduino IDE 1.8.13
Board v 1.8.2 Uno

The code below will write to the serial output unless Line 7 with "LK15" (or more values) is include in the initialization list. Then no serial output is produced. Best I can tell it is not exceeding memory limits. 

```
#include "ArduinoSTL.h"
//---------------------------------------------------------------------------------------------------------------------------
static const std::vector<std::string> noTxLocs { "IO78", "IO88", "IO77", "IO87", "IO76", "IO86", "IO75", "IO85", "IO84",
                                           "IO94", "IO83", "IO93", "IO82", "IO92", "JO02", "IO81", "IO91", "JO01",
                                           "IO70", "IO80", "IO90", "IO64", "PN31", "PN41", "PN20", "PN30", "PN40",
                                           "PM29", "PM39", "PM28", "PM38", "LK16",   
//                                           "LK15", // remove comment from this line
//                                           "LK14", // "LK13", // "LK23",
//                                           "LK24", "LK25", "LK26", "LK36", "LK35", "LK34", "LK33", "LK44", "LK45", 
//                                           "LK46", "LK47", "LK48", "LK58", "LK57", "LK56", "LK55"
};
//---------------------------------------------------------------------------------------------------------------------------
void setup() {
    Serial.begin(115200);
    std::cout  << "Setup running\n";
}
//---------------------------------------------------------------------------------------------------------------------------
void loop() {
}
```

tonk777, 2021-10-29:
Yes the amount of memory available is usually very small so this is not surprising. I wouldn't regard this as a bug - jut over ambition ;)

mike-matera, 2022-01-03:
Yes. This is probably a hardware limitation. Sorry!

closed 2022-01-03 by mike-matera