mike-matera/ArduinoSTL #66: Unexplainable error messages while compiling code using ArduinoSTL.h and array

salwaElkaddaoui, 2021-03-20:
I have written arduino C++ code in which I used std::array quite a lot. Of course I made sure to `#include<ArduinoSTL.h>` and `#include<array>`. The compiler raises multiple compilation errors (see below) without pointing to any line of the code:
`/home/salwa/Arduino/libraries/ArduinoSTL/src/del_opvs.cpp:25:53: error: 'std::size_t' has not been declared
 _UCXXEXPORT void operator delete[](void * ptr, std::size_t) throw(){
                                                     ^~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/del_ops.cpp:25:50: error: 'std::size_t' has not been declared
 _UCXXEXPORT void operator delete(void* ptr, std::size_t) throw(){
                                                  ^~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/del_opnt.cpp:25:56: error: 'nothrow_t' in namespace 'std' does not name a type
 _UCXXEXPORT void operator delete(void* ptr, const std::nothrow_t& ) throw() {
                                                        ^~~~~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_opnt.cpp:25:37: error: declaration of 'operator new' as non-function
 _UCXXEXPORT void* operator new(std::size_t numBytes, const std::nothrow_t& ) throw(){
                                     ^~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_opnt.cpp:25:37: error: 'size_t' is not a member of 'std'
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_opnt.cpp:25:37: note: suggested alternative:
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_handler.cpp:22:12: error: 'nothrow_t' in namespace 'std' does not name a type
 const std::nothrow_t std::nothrow = { };
            ^~~~~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_handler.cpp:25:6: error: 'new_handler' in namespace 'std' does not name a type
 std::new_handler __new_handler;
      ^~~~~~~~~~~
/home/salwa/Arduino/libraries/ArduinoSTL/src/new_handler.cpp:27:1: error: '_UCXXEXPORT' does not name a type
`
A similar issue was opened here [https://github.com/mike-matera/ArduinoSTL/issues/56](url), it was suggested to use #include"ArduinoSTL" instead of #include<ArduinoSTL>, I tried it but it didn't work, I tried downgrading the AVR board manager but that didn't solve the problem.

- Computer kernel/OS: 5.4.0-67-generic #75~18.04.1-Ubuntu
- ArduinoIDE: 1.8.13
- Tested AVR Boards manager: 1.8.1, 1.8.2, 1.8.3 (the problem persists with all these versions)
Any suggestions?

dasfdlinux, 2021-04-20:
Have you tried using Arduino 1.8.12? I'm also facing issues related to Arduino 1.8.13 and ArduinoSTL, and had to downgrade to 1.8.12 (hopefully temporarily).

salwaElkaddaoui, 2021-04-21:
@dasfdlinux No I didn't, I rewrote the code using c arrays, but thanks for the suggestion.

BCsabaEngine, 2021-07-04:
Waiting for... and thanks!

blackpancake, 2021-07-05:
The same issue here.
[image]

chrisflesher, 2021-07-22:
Same issue:
Arduino IDE version: 1.8.13
ArduinoSTL version: 1.1.0
Platform: Ubuntu 20.04

kleurbleur, 2021-07-30:
Some problem here.

Arduino IDE version: 1.8.15, 1.8.13, 1.8.12
ArduinoSTL version: 1.1.0
Platform: MacOS 10.15.7

ZhongKang97, 2021-09-20:
Same issue.

Arduino IDE version: 1.8.16
ArduinoSTL version: 1.1.0
Platform: Windows10 + VS2019
[image]

mike-matera, 2022-01-03:
It's fixed now! Thank you.

closed 2022-01-03 by mike-matera

BCsabaEngine, 2022-01-03:
We will try it. Thx!!!