mike-matera/ArduinoSTL #74: a lot of issues during compilation on plateformio / stm32 boards / arduino

camelator, 2022-03-07:
having a lot of compilation issues. Please see below my plateformio ini file and compilation errors

compilation errors:

from AruidoSTL/src/cmath:
there are no arguments to 'rand' that depend on a template parameter, so a declaration of 'rand' must be available [-fpermissive]
'::acos' has not been declared
'::asin' has not been declared
...

from ArduinoSTL/src/cstdlib:
'::abort' has not been declared
'::abs' has not been declared
...

etc...

plateformio.ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:blackpill_f411ce]
platform = ststm32
board = blackpill_f411ce 
board_build.mcu = stm32f411ceu6
framework = arduino
debug_tool = blackmagic
upload_protocol = blackmagic
monitor_speed = 115200

build_flags = 
  -D USER_SETUP_LOADED=1
  -include setup_tft.h

lib_deps =
# RECOMMENDED
# Accept new functionality in a backwards compatible manner and patches

  mike-matera/ArduinoSTL @ ^1.3.3 
  bodmer/TFT_eSPI @ ^2.4.42
  SPI
  ux

warnerjon12, 2022-03-10:
STM32 uses ARM Cortex-M, so you should be able to link against `libstdc++.a`. I see no reason to use ArduinoSTL for ARM based processors.

mike-matera, 2022-03-24:
Hello! @warnerjon12 is correct. You don't need or want this library on STM32.

closed 2022-03-24 by mike-matera