mike-matera/ArduinoSTL #77: Fails with latest IDE and Arduino

joeinaustin, 2022-04-20:
I see some references to this that are quite old.  It seemed to say this was fixed.  But I just upgraded top the latest of Arduino and the IDE and they have identical issues.  

(.text+0x0): multiple definition of `std::nothrow'

I have tried the 1.8.5 and the IDE rc6.   Am I missing something here, or has this issue resurfaced?

joeinaustin, 2022-04-20:
Downgrading AVR boards allowed me to continue.  Will there be a long term fix?

drsteveplatt, 2022-07-22:
Noted that it still fails on Arduino IDE 1.8.19.  VS and VSCode are also installed on the system.

The test program is an empty/new application with
```
#include <Arduino.STL>
using namespace std;
```
at the top.

Error message is (some blank lines removed)
```
Arduino: 1.8.19 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

new.cpp.o (symbol from plugin): In function `operator new(unsigned int)':
(.text+0x0): multiple definition of `std::nothrow'
libraries\ArduinoSTL\new_handler.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
```