mike-matera/ArduinoSTL #84: multiple definition of `std::nothrow'

jfjlaros, 2022-08-06:
When compiling the following sketch,

```cpp
#include <ArduinoSTL.h>

void setup() {}
void loop() {}
```

this error is raised:

```txt
arduino-cli compile --fqbn arduino:avr:pro --warnings all --output-dir build \
    --build-property compiler.cpp.extra_flags="-pedantic"
new.cpp.o (symbol from plugin): In function `operator new(unsigned int)':
(.text+0x0): multiple definition of `std::nothrow'
/tmp/arduino-sketch-DAE3E2C59CB46B47AC6BC234B0EB1886/libraries/ArduinoSTL/new_handler.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

Commenting out line 22 of `src/new_handler.cpp` seems to fix the issue, but this perhaps introduces other problems.

My set up:
- arduino-cli  Version: 0.25.1 Commit: 436f0bb9 Date: 2022-07-25T15:01:55Z
- ArduinoSTL version 1.3.3
- arduino:avr  core version 1.8.5

This issue can be reproduced in the [Wokwi simulator](https://wokwi.com/projects/339239665590600274).

POberauer, 2022-10-11:
downgrading to arduino:avr core version 1.8.3 fixed the issue.

POberauer, 2022-10-11:
or as @jfjlaros states,or remore / comment line 22 in  ../ArduinoSTL/src/new_handler.cpp:  const std::nothrow_t std::nothrow = { }; 
since for arduino core v1.8.5, std::nothrow, is defined in its ../core/arduino/new.cpp file.

rw3iss, 2022-12-05:
Was getting the same exact error. 
If I open the library file ArduinoSTL/src/new_handler.cpp and comment out line 22:
`// const std::nothrow_t std::nothrow = { };`
It will allow my program to compile.

quinn-n, 2022-12-29:
I'm getting the same error. The workaround posted by @rw3iss works.

crashback-exe, 2023-01-02:
same error here, fix by @rw3iss worked

mabsch, 2023-02-15:
same error, @rw3iss 's workaround worked.

odisseano, 2023-11-14:
Même erreur ;) Merci @rw3iss

reymalahay, 2023-11-29:
hi all,

is there going to be an official fix to this issue? i.e. a fix that does not require downgrading other libraries nor commenting out lines of code.

please advise.

thanks in advance,
rey malahay

mw66, 2024-01-20:
> Was getting the same exact error. If I open the library file ArduinoSTL/src/new_handler.cpp and comment out line 22: `// const std::nothrow_t std::nothrow = { };` It will allow my program to compile.

If this is not needed shall we create a PR to delete it, or guarded in some conditional MACRO?

```
#ifdef xxx
...
#endif
```

jfjlaros, 2024-01-20:
I have created PR #94 to implement the suggestion made in the first post of this issue.

sjimenez100, 2024-06-09:
> Was getting the same exact error. If I open the library file ArduinoSTL/src/new_handler.cpp and comment out line 22: `// const std::nothrow_t std::nothrow = { };` It will allow my program to compile.

Yep, that fixed it

wwh447, 2025-01-07:
same error, @rw3iss 's workaround worked.

clarityD, 2025-02-22:
TANGO YANKEE
BRAVO ZULU
I am CHARLIE MIKE
and OSCAR MIKE
grep -rl "const std::nothrow_t std::nothrow = { };" * | xargs sed -i '' 's|const std::nothrow_t std::nothrow = { };|// &|g'
MISSION SUCCESS