mike-matera/ArduinoSTL #95: Multiple definitions of `std::nothrow` on `__AVR_ARCH__`

zfields, 2024-03-05:
I'm not sure if this is the appropriate solution, but to allow an AVR project to compile correctly, I had to wrap the definition of `std::nothrow` in an `#ifdef`.

**`new_handler.cpp`**

```c++
#ifndef __AVR_ARCH__
	const std::nothrow_t std::nothrow = { };
#endif
```

Here is the compiler error for additional context:

```
/home/zak/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega2560 -o /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/ArduinoIoTCloud_Basic.ino.elf -Wl,--whole-archive /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/sketch/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoSTL/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoSTL/abi/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/cbor/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/cbor/lib/tinycbor/src/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/property/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/tls/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/tls/bearssl/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/tls/profile/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/tls/utility/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/utility/ota/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/utility/time/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/ArduinoIoTCloud/utility/watchdog/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Arduino_ConnectionHandler/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Blues_Wireless_Notecard/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Blues_Wireless_Notecard/note-c/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Wire/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Wire/utility/objs.a /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/libraries/Arduino_DebugUtils/objs.a -Wl,--no-whole-archive /home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build/../../../../../../../tmp/arduino/cores/arduino_avr_mega_97e1bcb735c3b776c955076fa7804c78/core.a -L/home/zak/Development/Arduino/generated_examples/ArduinoIoTCloud_Basic/build -lm
new.cpp.o (symbol from plugin): In function `operator new(unsigned int)':
(.text+0x0): multiple definition of `std::nothrow'
new_handler.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

thiagovmu, 2024-04-27:
Actually, I think that the fix is like below:
`const std::nothrow_t nothrow_t {};`

zfields, 2024-04-29:
Can you explain that in more detail? It looks like you are defining a new empty function.

I believe the purpose of the file is to define functions in the `std::` namespace.

thiagovmu, 2024-05-16:
By reading the code I understood that the intention is to get rid of writing std std all the time for nothrow, therefore, who wrote the code just made the mistake of duplicating the std word (the compiler error you saw).
The way I answered you can see that is just an initialization of nothrow variable where its type is std::nothrow.
Maybe it will be used somewhere else in the project but without the need of writing std::

AbbottF, 2024-05-22:
Thanks for the quick fix. Worked like a charm.